mirror/dd-wrt #13: why the project stop sync?

fkpwolf, 2015-03-02:
Looks it stopped sync after 2014. But there are many new changeset in 2015: http://svn.dd-wrt.com/timeline.

john-peterson, 2015-03-02:
the log is

	$ git push
	remote: error: GH001: Large files detected.
	remote: error: Trace: be522b856d9ca824b6cc8c900db1aaec
	remote: error: See http://git.io/iEPt8g for more information.
	remote: error: File src/linux/universal/linux-3.18/3.18-rc4.diff is 112.51 MB; this exceeds GitHub's file size limit of 100 MB      
	To git@github.com:mirror/dd-wrt.git
	 ! [remote rejected] master -> master (pre-receive hook declined)
	error: failed to push some refs to 'git@github.com:mirror/dd-wrt.git'

what should i do?

fkpwolf, 2015-03-02:
I see. Why does dd-wrt have so big file? really bad.

john-peterson, 2015-03-02:
i tried bfg and it still fails

	user@ubuntu:/mnt/lv/repo/dd-wrt$ java -jar ~/bin/bfg.jar -b 100M

	Using repo : /mnt/lv/repo/dd-wrt/.git

	Scanning packfile for large blobs: 995485
	Scanning packfile for large blobs completed in 876,975 ms.
	Found 1 blob ids for large blobs - biggest=117974818 smallest=117974818
	Total size (unpacked)=117974818
	Found 321840 objects to protect
	Found 4 commit-pointing refs : HEAD, refs/heads/master, refs/remotes/git-svn, refs/remotes/origin/master

	Protected commits
	-----------------

	These are your protected commits, and so their contents will NOT be altered:

	 * commit 0af2263a (protected by 'HEAD') - contains 1 dirty file :
		- src/linux/universal/linux-3.18/3.18-rc4.diff (112.5 MB)

	WARNING: The dirty content above may be removed from other commits, but as
	the *protected* commits still use it, it will STILL exist in your repository.

	Details of protected dirty content have been recorded here :

	/mnt/lv/repo/dd-wrt.bfg-report/2015-03-02/15-02-27/protected-dirt/

	If you *really* want this content gone, make a manual commit that removes it,
	and then run the BFG on a fresh copy of your repo.

	Cleaning
	--------

	Found 26405 commits
	Cleaning commits:       100% (26405/26405)
	Cleaning commits completed in 36,125 ms.

	Updating 2 Refs
	---------------

		Ref                    Before     After
		------------------------------------------
		refs/heads/master    | 0af2263a | 0ceab42c
		refs/remotes/git-svn | 0af2263a | 0ceab42c

	Updating references:    100% (2/2)
	...Ref update completed in 177 ms.

	Commit Tree-Dirt History
	------------------------

		Earliest                                              Latest
		|                                                          |
		.........................................................DDD

		D = dirty commits (file tree fixed)
		m = modified commits (commit message or parents changed)
		. = clean commits (no changes to file tree)

								Before     After
		-------------------------------------------
		First modified commit | 686d1208 | f1bac602
		Last dirty commit     | 6f5f36fe | 2302897a

	Deleted files
	-------------

		Filename        Git id
		-----------------------------------
		3.18-rc4.diff | 18747a61 (112.5 MB)

	In total, 3759 object ids were changed. Full details are logged here:

		/mnt/lv/repo/dd-wrt.bfg-report/2015-03-02/15-02-27

	BFG run is complete! When ready, run: git reflog expire --expire=now --all && git gc --prune=now --aggressive

	Has the BFG saved you time?  Support the BFG on BountySource:  https://j.mp/fund-bfg

	user@ubuntu:/mnt/lv/repo/dd-wrt$ git svn rebase
	Partial-rebuilding .git/svn/refs/remotes/git-svn/.rev_map.52c4871e-980c-0410-b1e0-e73912ce01f8 ...
	Currently at 26405 = 0af2263ab483058b81b43c257c8fb4cefacf4364
	...
	Done rebuilding .git/svn/refs/remotes/git-svn/.rev_map.52c4871e-980c-0410-b1e0-e73912ce01f8
		M   src/router/httpd/visuals/wireless_broadcom.c
	r26406 = 26a39461605b8b1975788c49023169693da31835 (refs/remotes/git-svn)
		M   src/router/kromo/dd-wrt/Wireless_WDS.asp
	...
	r26424 = 7fdf0af21aaf08ab3bb01f8114ac74ae9fd9dda9 (refs/remotes/git-svn)
	Auto packing the repository in background for optimum performance.
	See "git help gc" for manual housekeeping.
	First, rewinding head to replay your work on top of it...
	Fast-forwarded master to refs/remotes/git-svn.

	user@ubuntu:/mnt/lv/repo/dd-wrt$ git push
	Counting objects: 76279, done.
	Compressing objects: 100% (55486/55486), done.
	Writing objects: 100% (76279/76279), 183.10 MiB | 1.40 MiB/s, done.
	Total 76279 (delta 43147), reused 53308 (delta 20268)
	remote: error: GH001: Large files detected.
	remote: error: Trace: 01adc0861745bd55e3387edce13e7ca7
	remote: error: See http://git.io/iEPt8g for more information.
	remote: error: File src/linux/universal/linux-3.18/3.18-rc4.diff is 112.51 MB; this exceeds GitHub's file size limit of 100 MB
	To git@github.com:mirror/dd-wrt.git
	 ! [remote rejected] master -> master (pre-receive hook declined)
	error: failed to push some refs to 'git@github.com:mirror/dd-wrt.git'

john-peterson, 2015-03-02:
how can i remove it without "git svn rebase" adding it again?

fkpwolf, 2015-03-02:
I am not sure. But I think you can ignore the big diff file. I would be a temp file for backup or other use.

john-peterson, 2015-03-02:
the problem was that bfg staged the deleted file. the bad commit 686d1208 is changed to f1bac602 and the mirror should work now

	user@ubuntu:~/repo/dd-wrt$ java -jar ~/bin/bfg.jar -b 100M --no-blob-protection

	Using repo : /home/user/repo/dd-wrt/.git

	Scanning packfile for large blobs: 995485
	Scanning packfile for large blobs completed in 623,695 ms.
	Found 1 blob ids for large blobs - biggest=117974818 smallest=117974818
	Total size (unpacked)=117974818
	Found 0 objects to protect
	Found 4 commit-pointing refs : HEAD, refs/heads/master, refs/remotes/git-svn, refs/remotes/origin/master

	Protected commits
	-----------------

	You're not protecting any commits, which means the BFG will modify the contents of even *current* commits.

	This isn't recommended - ideally, if your current commits are dirty, you should fix up your working copy and commit that, check that your build still works, and only then run the BFG to clean up your history.

	Cleaning
	--------

	Found 26405 commits
	Cleaning commits:       100% (26405/26405)
	Cleaning commits completed in 52,601 ms.

	Updating 2 Refs
	---------------

		Ref                    Before     After
		------------------------------------------
		refs/heads/master    | 0af2263a | c32245f5
		refs/remotes/git-svn | 0af2263a | c32245f5

	Updating references:    100% (2/2)
	...Ref update completed in 92 ms.

	Commit Tree-Dirt History
	------------------------

		Earliest                                              Latest
		|                                                          |
		.........................................................DDD

		D = dirty commits (file tree fixed)
		m = modified commits (commit message or parents changed)
		. = clean commits (no changes to file tree)

								Before     After
		-------------------------------------------
		First modified commit | 686d1208 | f1bac602
		Last dirty commit     | 0af2263a | c32245f5

	Deleted files
	-------------

		Filename        Git id
		-----------------------------------
		3.18-rc4.diff | 18747a61 (112.5 MB)

	In total, 3761 object ids were changed. Full details are logged here:

		/home/user/repo/dd-wrt.bfg-report/2015-03-02/17-55-23

	BFG run is complete! When ready, run: git reflog expire --expire=now --all && git gc --prune=now --aggressive

	Has the BFG saved you time?  Support the BFG on BountySource:  https://j.mp/fund-bfg

	user@ubuntu:~/repo/dd-wrt$ git stat
	On branch master
	Changes to be committed:
	  (use "git reset HEAD <file>..." to unstage)

		new file:   src/linux/universal/linux-3.18/3.18-rc4.diff
		deleted:    src/linux/universal/linux-3.18/3.18-rc4.diff.REMOVED.git-id

	Untracked files not listed (use -u option to show untracked files)

	user@ubuntu:~/repo/dd-wrt$ git checkout -f

	user@ubuntu:~/repo/dd-wrt$ git stat
	On branch master
	nothing to commit (use -u to show untracked files)

	user@ubuntu:~/repo/dd-wrt$ git svn rebase
	Partial-rebuilding .git/svn/refs/remotes/git-svn/.rev_map.52c4871e-980c-0410-b1e0-e73912ce01f8 ...
	Currently at 26405 = 0af2263ab483058b81b43c257c8fb4cefacf4364
	r25370 = f1bac602c817709ac3a7e4424532bb1e3324c6f2
	...
	r26405 = c32245f5fe281c22cfa8be9fbffb456a7f1f1e62
	Done rebuilding .git/svn/refs/remotes/git-svn/.rev_map.52c4871e-980c-0410-b1e0-e73912ce01f8
	...
	r26424 = 61d07fb4b8bb8deceaef756c34b3ab774a7e24eb (refs/remotes/git-svn)
	Auto packing the repository in background for optimum performance.
	See "git help gc" for manual housekeeping.
	First, rewinding head to replay your work on top of it...
	Fast-forwarded master to refs/remotes/git-svn.
	Auto packing the repository in background for optimum performance.
	See "git help gc" for manual housekeeping.
	creating empty directory: ar5315_microredboot/microredboot/ecos/doc/sgml/tutorials/pix
	...

	user@ubuntu:~/repo/dd-wrt$ git push
	Counting objects: 76275, done.
	Compressing objects: 100% (55480/55480), done.
	Writing objects: 100% (76275/76275), 159.27 MiB | 1.33 MiB/s, done.
	Total 76275 (delta 43144), reused 53309 (delta 20270)
	To git@github.com:mirror/dd-wrt.git
	   0cd51b5..61d07fb  master -> master

closed 2015-03-02 by john-peterson

fkpwolf, 2015-03-03:
cool! thanks a lot!

realdreams, 2015-07-17:
What do you continue syncing the svn after using bfg?

After git svn fetch I can't do another git svn rebase

$ git svn rebase
Cannot rebase with uncommitted changes:
On branch master
Your branch is up-to-date with 'origin/master'.
Changes to be committed:
  (use "git reset HEAD <file>..." to unstage)

        new file:   src/linux/universal/linux-3.18/3.18-rc4.diff
        deleted:    src/linux/universal/linux-3.18/3.18-rc4.diff.REMOVED.git-id

john-peterson, 2015-07-17:
compare the commands at the top of my posts. the difference is "git checkout -f"