mirror/dd-wrt #16: migration issue

asdfljdasflkjwfd, 2015-12-25:
I am migrating the repository (Debian GNU/Linux 8.2 (jessie), git version 2.1.4) to a new host (Ubuntu 12.04.5 LTS git version 2.6.4 compiled from source, same result with git version 1.7.9.5 shipped with the distribution).

I used `rsync -azP` to transfer the entire repository, but on the destination I get uncommitted changes

```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        deleted:    src/linux/adm5120/linux-2.6.23/arch/blackfin/lib/modsi3.S
        deleted:    src/linux/adm5120/linux-2.6.23/include/asm-parisc/posix_types.h
        modified:   src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj
        modified:   src/router/openvpn/openvpn.sln
        modified:   src/router/openvpn/src/compat/compat.vcxproj
        modified:   src/router/openvpn/src/compat/compat.vcxproj.filters
        modified:   src/router/openvpn/src/openvpn/openvpn.vcxproj
        modified:   src/router/openvpn/src/openvpn/openvpn.vcxproj.filters
        modified:   src/router/openvpn/src/openvpnserv/openvpnserv.vcxproj
        modified:   src/router/openvpn/src/openvpnserv/openvpnserv.vcxproj.filters
        modified:   src/router/php5/README.input_filter
        modified:   src/router/php5/Zend/RFCs/002.txt
        modified:   src/router/php5/Zend/RFCs/003.txt
        modified:   src/router/php5/ext/dba/libcdb/cdb.c
        modified:   src/router/php5/ext/dba/libcdb/cdb_make.c
        modified:   src/router/php5/ext/dba/libflatfile/flatfile.c
        modified:   src/router/php5/ext/dba/libinifile/inifile.c
        modified:   src/router/php5/ext/enchant/enchant.c
        modified:   src/router/php5/ext/exif/exif.c
        modified:   src/router/php5/ext/filter/filter.c
        modified:   src/router/php5/ext/iconv/php_iconv.h
        modified:   src/router/php5/ext/ldap/ldap.c
        modified:   src/router/php5/ext/mysqlnd/mysqlnd.h
        modified:   src/router/php5/ext/oci8/oci8.c
        modified:   src/router/php5/ext/pdo_pgsql/pdo_pgsql.c
        modified:   src/router/php5/ext/phar/phar.c
        modified:   src/router/php5/ext/phar/phar/pharcommand.inc
        modified:   src/router/php5/ext/posix/posix.c
        modified:   src/router/php5/ext/recode/recode.c
        modified:   src/router/php5/ext/reflection/php_reflection.c
        modified:   src/router/php5/ext/simplexml/simplexml.c
        modified:   src/router/php5/ext/skeleton/create_stubs
        modified:   src/router/php5/ext/sysvmsg/sysvmsg.c
        modified:   src/router/php5/ext/tidy/tidy.c
        modified:   src/router/php5/ext/zip/php_zip.c
        modified:   src/router/php5/run-tests.php
        modified:   src/router/php5/sapi/continuity/capi.c
        modified:   src/router/php5/sapi/nsapi/nsapi.c

It took 2.95 seconds to enumerate untracked files. 'status -uno'
may speed it up, but you have to be careful not to forget to add
new files yourself (see 'git help status').
no changes added to commit (use "git add" and/or "git commit -a")
```

md5sum of these files match those on the original host

> c30df7f4b27816dd8cfc64d4a4cc84d2  src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj
> c30df7f4b27816dd8cfc64d4a4cc84d2  src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj

another pass with `rsync -avH --delete` did not fix this issue

Is it related to https://stackoverflow.com/questions/26542226/git-showing-identical-files-as-changed?
I got the same issue with tar.gz transferred over HTTP

> 63a656fa823450b4d4cec14d73ac7d9ff1329246  src/router/openvpn/openvpn.sln
> 63a656fa823450b4d4cec14d73ac7d9ff1329246  src/router/openvpn/openvpn.sln

> src/router/openvpn/openvpn.sln: UTF-8 Unicode (with BOM) text, with CRLF line terminators
> src/router/openvpn/openvpn.sln: UTF-8 Unicode (with BOM) text, with CRLF line terminators

>>>>>>

Since git reset --hard is quite cpu intensive, I used git checkout on every file in modified state. After 1 pass I still got 5 files showing modified. I did another pass on the remaining files, 2 left. Then did it again. Now I have a clean repository.

Any thoughts on what the issue might be?

The mirroring process has been resumed on the new host.

@mirror/write-access

asdfljdasflkjwfd, 2016-01-01:
I did another migration to a different host and got the exact same thing. I ended up using git reset --hard to fix it. Still no clue what's causing it.

TimothyGu, 2016-01-02:
Huh. According to the SO post the mtime might have something to do with it, so maybe rsync should be invoked with `-t` which preserves it?

TimothyGu, 2016-01-02:
Also, have you tried to manually compare the files between the two machines?

asdfljdasflkjwfd, 2016-01-03:
Looks like an encoding issue.
```
$ git diff
diff --git a/src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj b/src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj
index e6491ab..8b7ec22 100644
--- a/src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj
+++ b/src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj
@@ -1,69 +1,69 @@
-<U+FEFF><?xml version="1.0" encoding="utf-8"?>
-<Project DefaultTargets="Build" ToolsVersion="4.0" xmlns="http://schemas.microsoft.com/developer/msbuild/2003">
-  <ItemGroup Label="ProjectConfigurations">
-    <ProjectConfiguration Include="Debug|Win32">
-      <Configuration>Debug</Configuration>
-      <Platform>Win32</Platform>
-    </ProjectConfiguration>
-    <ProjectConfiguration Include="Release|Win32">
-      <Configuration>Release</Configuration>
-      <Platform>Win32</Platform>
-    </ProjectConfiguration>
-  </ItemGroup>
-  <PropertyGroup Label="Globals">
-    <ProjectGuid>{8598C2C8-34C4-47A1-99B0-7C295A890615}</ProjectGuid>
-    <RootNamespace>msvc-generate</RootNamespace>
-    <Keyword>MakeFileProj</Keyword>
-  </PropertyGroup>
-  <Import Project="$(VCTargetsPath)\Microsoft.Cpp.Default.props" />
-  <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Release|Win32'" Label="Configuration">
-    <ConfigurationType>Makefile</ConfigurationType>
-  </PropertyGroup>
-  <PropertyGroup Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'" Label="Configuration">
-    <ConfigurationType>Makefile</ConfigurationType>
-  </PropertyGroup>
-  <Import Project="$(VCTargetsPath)\Microsoft.Cpp.props" />
-  <ImportGroup Label="ExtensionSettings">
-  </ImportGroup>
-  <ImportGroup Condition="'$(Configuration)|$(Platform)'=='Release|Win32'" Label="PropertySheets">
-    <Import Project="$(UserRootDir)\Microsoft.Cpp.$(Platform).user.props" Condition="exists('$(UserRootDir)\Microsoft.Cpp.$(Platform).user.props')
-  </ImportGroup>
-  <ImportGroup Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'" Label="PropertySheets">
-    <Import Project="$(UserRootDir)\Microsoft.Cpp.$(Platform).user.props" Condition="exists('$(UserRootDir)\Microsoft.Cpp.$(Platform).user.props')
-  </ImportGroup>
-  <PropertyGroup Label="UserMacros" />
-  <PropertyGroup>
-    <_ProjectFileVersion>10.0.30319.1</_ProjectFileVersion>
-    <OutDir Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">$(Configuration)\</OutDir>
-    <IntDir Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">$(Configuration)\</IntDir>
-    <NMakeBuildCommandLine Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">nmake -f Makefile.mak all</NMakeBuildCommandLine>
-    <NMakeReBuildCommandLine Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">nmake -f Makefile.mak clean all</NMakeReBuildCommandLine>
-    <NMakeCleanCommandLine Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">nmake -f Makefile.mak clean</NMakeCleanCommandLine>
-    <NMakeOutput Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">config-msvc-version.h</NMakeOutput>
-    <NMakePreprocessorDefinitions Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">WIN32;_DEBUG;$(NMakePreprocessorDefinitions)</NMakePre
-    <NMakeIncludeSearchPath Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">$(NMakeIncludeSearchPath)</NMakeIncludeSearchPath>
-    <NMakeForcedIncludes Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">$(NMakeForcedIncludes)</NMakeForcedIncludes>
-    <NMakeAssemblySearchPath Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">$(NMakeAssemblySearchPath)</NMakeAssemblySearchPath>
-    <NMakeForcedUsingAssemblies Condition="'$(Configuration)|$(Platform)'=='Debug|Win32'">$(NMakeForcedUsingAssemblies)</NMakeForcedUsingAssemblie
-    <OutDir Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(Configuration)\</OutDir>
-    <IntDir Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(Configuration)\</IntDir>
-    <NMakeBuildCommandLine Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">nmake -f Makefile.mak all</NMakeBuildCommandLine>
-    <NMakeReBuildCommandLine Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">nmake -f Makefile.mak clean all</NMakeReBuildCommandLine>
-    <NMakeCleanCommandLine Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">nmake -f Makefile.mak clean</NMakeCleanCommandLine>
-    <NMakeOutput Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">config-msvc-version.h</NMakeOutput>
-    <NMakePreprocessorDefinitions Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">WIN32;NDEBUG;$(NMakePreprocessorDefinitions)</NMakeP
-    <NMakeIncludeSearchPath Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(NMakeIncludeSearchPath)</NMakeIncludeSearchPath>
-    <NMakeForcedIncludes Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(NMakeForcedIncludes)</NMakeForcedIncludes>
-    <NMakeAssemblySearchPath Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(NMakeAssemblySearchPath)</NMakeAssemblySearchPath>
-    <NMakeForcedUsingAssemblies Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(NMakeForcedUsingAssemblies)</NMakeForcedUsingAssembl
-  </PropertyGroup>
-  <ItemDefinitionGroup>
-  </ItemDefinitionGroup>
-  <ItemGroup>
[... 53 lines cut ...]
+    <OutDir Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(Configuration)\</OutDir>
[... cut ...]
+    <NMakeAssemblySearchPath Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(NMakeAssemblySearchPath)</NMakeAssemblySearchPath>
+    <NMakeForcedUsingAssemblies Condition="'$(Configuration)|$(Platform)'=='Release|Win32'">$(NMakeForcedUsingAssemblies)</NMakeForcedUsingAssembl
+  </PropertyGroup>
+  <ItemDefinitionGroup>
+  </ItemDefinitionGroup>
+  <ItemGroup>
+    <None Include="Makefile.mak" />
+    <None Include="msc-generate.js" />
+  </ItemGroup>
+  <Import Project="$(VCTargetsPath)\Microsoft.Cpp.targets" />
+  <ImportGroup Label="ExtensionTargets">
+  </ImportGroup>
 </Project>
\ No newline at end of file

```

asdfljdasflkjwfd, 2016-01-07:
mtime and permissions are actually the same. But git still thinks the file was entirely changed.

jkldgoefgkljefogeg, 2016-01-19:
Today `git svn rebase` failed (the cron job) caused by those exact files. I never figured out why these files are so problematic

$ git svn rebase
src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj: needs update
src/router/openvpn/openvpn.sln: needs update
src/router/openvpn/src/compat/compat.vcxproj: needs update
src/router/openvpn/src/compat/compat.vcxproj.filters: needs update
src/router/openvpn/src/openvpn/openvpn.vcxproj: needs update
src/router/openvpn/src/openvpn/openvpn.vcxproj.filters: needs update
src/router/openvpn/src/openvpnserv/openvpnserv.vcxproj: needs update
src/router/openvpn/src/openvpnserv/openvpnserv.vcxproj.filters: needs update
src/router/php5/README.input_filter: needs update
src/router/php5/Zend/RFCs/002.txt: needs update
src/router/php5/Zend/RFCs/003.txt: needs update
src/router/php5/ext/dba/libcdb/cdb.c: needs update
src/router/php5/ext/dba/libcdb/cdb_make.c: needs update
src/router/php5/ext/dba/libflatfile/flatfile.c: needs update
src/router/php5/ext/dba/libinifile/inifile.c: needs update
src/router/php5/ext/enchant/enchant.c: needs update
src/router/php5/ext/exif/exif.c: needs update
src/router/php5/ext/filter/filter.c: needs update
src/router/php5/ext/iconv/php_iconv.h: needs update
src/router/php5/ext/ldap/ldap.c: needs update
src/router/php5/ext/mysqlnd/mysqlnd.h: needs update
src/router/php5/ext/oci8/oci8.c: needs update
src/router/php5/ext/pdo_pgsql/pdo_pgsql.c: needs update
src/router/php5/ext/phar/phar.c: needs update
src/router/php5/ext/phar/phar/pharcommand.inc: needs update
src/router/php5/ext/posix/posix.c: needs update
src/router/php5/ext/recode/recode.c: needs update
src/router/php5/ext/reflection/php_reflection.c: needs update
src/router/php5/ext/simplexml/simplexml.c: needs update
src/router/php5/ext/skeleton/create_stubs: needs update
src/router/php5/ext/sysvmsg/sysvmsg.c: needs update
src/router/php5/ext/tidy/tidy.c: needs update
src/router/php5/ext/zip/php_zip.c: needs update
src/router/php5/run-tests.php: needs update
src/router/php5/sapi/continuity/capi.c: needs update
src/router/php5/sapi/nsapi/nsapi.c: needs update
update-index --refresh: command returned error: 1

$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   src/router/openvpn/build/msvc/msvc-generate/msvc-generate.vcxproj
        modified:   src/router/openvpn/openvpn.sln
        modified:   src/router/openvpn/src/compat/compat.vcxproj
        modified:   src/router/openvpn/src/compat/compat.vcxproj.filters
        modified:   src/router/openvpn/src/openvpn/openvpn.vcxproj
        modified:   src/router/openvpn/src/openvpn/openvpn.vcxproj.filters
        modified:   src/router/openvpn/src/openvpnserv/openvpnserv.vcxproj
        modified:   src/router/openvpn/src/openvpnserv/openvpnserv.vcxproj.filters
        modified:   src/router/php5/README.input_filter
        modified:   src/router/php5/Zend/RFCs/002.txt
        modified:   src/router/php5/Zend/RFCs/003.txt
        modified:   src/router/php5/ext/dba/libcdb/cdb.c
        modified:   src/router/php5/ext/dba/libcdb/cdb_make.c
        modified:   src/router/php5/ext/dba/libflatfile/flatfile.c
        modified:   src/router/php5/ext/dba/libinifile/inifile.c
        modified:   src/router/php5/ext/enchant/enchant.c
        modified:   src/router/php5/ext/exif/exif.c
        modified:   src/router/php5/ext/filter/filter.c
        modified:   src/router/php5/ext/iconv/php_iconv.h
        modified:   src/router/php5/ext/ldap/ldap.c
        modified:   src/router/php5/ext/mysqlnd/mysqlnd.h
        modified:   src/router/php5/ext/oci8/oci8.c
        modified:   src/router/php5/ext/pdo_pgsql/pdo_pgsql.c
        modified:   src/router/php5/ext/phar/phar.c
        modified:   src/router/php5/ext/phar/phar/pharcommand.inc
        modified:   src/router/php5/ext/posix/posix.c
        modified:   src/router/php5/ext/recode/recode.c
        modified:   src/router/php5/ext/reflection/php_reflection.c
        modified:   src/router/php5/ext/simplexml/simplexml.c
        modified:   src/router/php5/ext/skeleton/create_stubs
        modified:   src/router/php5/ext/sysvmsg/sysvmsg.c
        modified:   src/router/php5/ext/tidy/tidy.c
        modified:   src/router/php5/ext/zip/php_zip.c
        modified:   src/router/php5/run-tests.php
        modified:   src/router/php5/sapi/continuity/capi.c
        modified:   src/router/php5/sapi/nsapi/nsapi.c

Had to do a `git reset --hard`,

TimothyGu, 2016-01-19:
Is it because of the line ending?

jkldgoefgkljefogeg, 2016-01-19:
All lines I have checked used LF. I just added core.autocrlf=input. These files have no recent changes on svn.

closed 2016-10-09 by jkldgoefgkljefogeg