mirror/dd-wrt #21: Support IPSET

houzi-, 2016-04-19:
I compile "xt_set.ko" kernel module. 

 In ".config_bcmmips" file of ddwrt svn 3.10 kernel directory: 
```
 ---------------------- 
 # 
 # Xtables combined modules 
 # 
 CONFIG_NETFILTER_XT_MARK=y 
 CONFIG_NETFILTER_XT_CONNMARK=y 
 # CONFIG_NETFILTER_XT_SET is not set 
 ..... 
 CONFIG_IP_SET=y 
 CONFIG_IP_SET_MAX=256 
 CONFIG_IP_SET_BITMAP_IP=y 
 CONFIG_IP_SET_BITMAP_IPMAC=y 
 CONFIG_IP_SET_BITMAP_PORT=y 
 CONFIG_IP_SET_HASH_IP=y 
 CONFIG_IP_SET_HASH_IPPORT=y 
 CONFIG_IP_SET_HASH_IPPORTIP=y 
 CONFIG_IP_SET_HASH_IPPORTNET=y 
 CONFIG_IP_SET_HASH_NET=y 
 CONFIG_IP_SET_HASH_NETPORT=y 
 # CONFIG_IP_SET_HASH_NETIFACE is not set 
 CONFIG_IP_SET_LIST_SET=y 
 # CONFIG_IP_VS is not set 
 ---------------------- 
```
Load modules but the error:
```
insmod: cannot insert 'xt_set.ko': unknown symbol in module". 
```
Can you help me?

houzi-, 2016-05-01:
(R6300V1)
Compiled into the kernel module does not work , are mipsel platform like this ! armv7 platform no problem
```
Jan  1 00:07:56 DD-WRT kern.warn kernel: xt_set: Unknown symbol ip_set_add (err 0)
Jan  1 00:07:56 DD-WRT kern.warn kernel: xt_set: Unknown symbol ip_set_test (err 0)
Jan  1 00:07:56 DD-WRT kern.warn kernel: xt_set: Unknown symbol ip_set_del (err 0)
Jan  1 00:07:56 DD-WRT kern.warn kernel: xt_set: Unknown symbol ip_set_nfnl_put (err 0)
Jan  1 00:07:56 DD-WRT kern.warn kernel: xt_set: Unknown symbol ip_set_nfnl_get_byindex (err 0)
```

GXTX, 2016-05-09:
This repo is just a mirror, I'd report this to: http://svn.dd-wrt.com/

troyliu0105, 2018-02-22:
@houzi- Hey, did you remember how did you solve this? I met the same problem now...

BrainSlayer, 2018-10-19:
i have no use for ipset right now in dd-wrt itself due the restricted space only required features are compiled in, but you're free you make your own variant with the feature sets you want.

BrainSlayer, 2018-10-19:
its not enough to just compile the module, you need to recompile the kernel since these required symbols are always compiled direct into the kernel without any module support

BrainSlayer, 2018-10-19:
correction. there is a module named "ip_set.ko" this must be loaded first

houzi-, 2018-10-25:
@BrainSlayer OK！thanks  I try