mirror/dd-wrt #22: Mirror not updated for about 8 days now

sdelafond, 2016-12-21:
See http://svn.dd-wrt.com/timeline for recent changes.

jkldgoefgkljefogeg, 2016-12-21:
Thanks for the heads up. After a migration some files got "dirty", I will see if this happens again. Never figured out what the cause was.

`$ git svn rebase
src/router/openvpn/src/compat/compat.vcxproj: needs update
src/router/openvpn/src/compat/compat.vcxproj.filters: needs update
src/router/openvpn/src/openvpn/openvpn.vcxproj: needs update
src/router/openvpn/src/openvpn/openvpn.vcxproj.filters: needs update
src/router/openvpn/src/openvpnserv/openvpnserv.vcxproj: needs update
update-index --refresh: command returned error: 1

$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   src/router/openvpn/src/compat/compat.vcxproj
        modified:   src/router/openvpn/src/compat/compat.vcxproj.filters
        modified:   src/router/openvpn/src/openvpn/openvpn.vcxproj
        modified:   src/router/openvpn/src/openvpn/openvpn.vcxproj.filters
        modified:   src/router/openvpn/src/openvpnserv/openvpnserv.vcxproj

no changes added to commit (use "git add" and/or "git commit -a")`

sdelafond, 2016-12-21:
Very weird, especially in a "read-only" clone like the one you're maintaining. I've never seen that before either.

jkldgoefgkljefogeg, 2016-12-22:
No more issue so far.

closed 2016-12-24 by sdelafond