mirror/dd-wrt #23: mirroring stopped

jkldgoefgkljefogeg, 2017-05-15:
This repository was missed during an earlier migration. The original files are now lost and I am unable to resume it, keep getting `error: git-svn died of signal 11`. @TimothyGu would you like to take over this mirror? Thanks.

TimothyGu, 2017-05-18:
I don't think I have the resources to continue this mirror, I'm afraid D:

sdelafond, 2017-05-23:
Should we just consider this mirror as dead once and for all, then ? If so, it'd probably be a good idea to clearly state it in the README, or to remove it entirely.

jkldgoefgkljefogeg, 2017-05-23:
Wouldn't adding README.md make it impossible for further commits? Judging from the watch/star/fork count, it is not entirely useless. I suggest just leaving it as it is and perhaps someone has interest to pick it back up later.

sdelafond, 2017-05-24:
Sure, so why not just change the *description* of the project to clearly mention it's not been updated for months ? I mean, this mirror is forked/starred because it's meant to be an *up-to-date* view of the official dd-wrt repo. A 3 month-old snapshot will not do much good in that regard.

jkldgoefgkljefogeg, 2017-05-24:
Description updated, thanks.

Neustradamus, 2017-10-05:
Can you relaunch it?

closed 2018-03-30 by BrainSlayer