mirror/dd-wrt #24: request for hosting the mirror

BrainSlayer, 2017-12-28:
i'm the author and maintainer of dd-wrt. if you would like to give the job into my hand, just do it

jkldgoefgkljefogeg, 2017-12-30:
That would be great if you can host an official github repo. @john-peterson can add you to the org.

GXTX, 2018-03-20:
*cough*

jkldgoefgkljefogeg, 2018-03-20:
poke @john-peterson

BrainSlayer, 2018-03-20:
the problem is that github has a repo quota even if github itself states that there is no quota. i already tried to clone our svn server, but it failed since the repositoriy size was to big

jkldgoefgkljefogeg, 2018-03-20:
I had to use BFG to remove some large blobs from previous commits (current repo doesn't have those large blobs any more). Github has a hard 100MB per file limit.
https://help.github.com/articles/what-is-my-disk-quota/

john-peterson, 2018-03-20:
Invitation sent

BrainSlayer, 2018-03-20:
joined

closed 2018-03-30 by BrainSlayer