mirror/dd-wrt #29: Support for Archer C9 v5

Pelek7, 2019-12-18:
Like in the subject, is there any possibility to make the firmware also for the version 5 (there's firmware for all vesrions below)

BrainSlayer, 2020-04-20:
depends if i get such a device for research

kernel-panic69, 2022-06-13:
Currently, all but the v5 are officially supported:
https://forum.dd-wrt.com/wiki/index.php/TP_Link_Archer_C9
https://forum.dd-wrt.com/wiki/index.php/Supported_Devices#TP-Link
Last I knew, the v5 was still a no-go / WIP:
https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=315680

jdpc86, 2022-07-01:
Hmm not much progress here with tplink archer c9 v5 still 

发自我的iPhone

kernel-panic69, 2022-07-29:
This is the last information in hand here regarding this device.  If it happens, it will be when BrainSlayer finds the mood to finish the port.  He has a donated test device in hand.  It would be helpful if we had more development collaboration to add device support, but the community is lacking skilled technical artisans who have the expertise in reverse engineering devices and porting them to the firmware.  Best advice here is to be patient.  

[TP-Link Archer C9 (EU) V5 DD-WRT Support (NO GO)](https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=315680&postdays=0&postorder=asc&start=63)

From: 
Date: Sat, Nov 20, 2021 at 1:06 PM
Subject: Re: Archer C9 v5 revisited
To: Sebastian Gottschall <[s.gottschall@dd-wrt.com](mailto:s.gottschall@dd-wrt.com)>

And it seems folks are adamant about supporting this device:

[TPLINK Archer C9 v5 unlock the protection](https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=330769)

As I explained to the user, you already have this information.... 

On Sun, Jul 18, 2021 at 5:27 PM Sebastian Gottschall <[s.gottschall@dd-wrt.com](mailto:s.gottschall@dd-wrt.com)> wrote:
i broke the cryptoshit already a while ago but never had physical access
to the device for playing with my findings

Am 18.07.2021 um 23:56 schrieb :
> Someone else is insistent on asking questions. Apparently, there is an
> unsigned beta version of stock firmware that was posted in their
> forums.  Not sure if this would help add this device or not, but I
> know that your last response was that is was not going to happen.
>
> [a bit of a question about Archer C9 v5...](https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=329699)
>
> /R,
> kp69

jdpc86, 2022-08-29:
...hey Sebastian...did you break the crypto? we can connect and I can test it out :)

Jindong.

lscorcia, 2023-03-12:
A new firmware is out for C9 v5: https://www.tp-link.com/en/support/download/archer-c9/#Firmware

It seems to mainly fix https://nvd.nist.gov/vuln/detail/CVE-2021-45608 . Unfortunately it seems any exploit is going to be difficult (on my router, port 20005 isn't listening. Maybe some router feature has to be enabled first?).