mirror/dd-wrt #35: Instructions how to build the ndpi-netfilter

houmie, 2020-09-29:
Hello,

It's amazing to see that ndpi-netfilter is here patched and seems to be maintained.

I'm trying this right now on Ubuntu 20.04 with Kernel: 
```Linux de-fsn-x 5.4.0-1025-aws #25-Ubuntu SMP Fri Sep 11 09:37:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux```

After going to dd-wrt/tree/master/src/router/ndpi-netfilter/ndpi-netfilter
I run the following commands:

chmod +x ./autogen.sh
./autogen.sh
cd src/lib
make ndpi_network_list.c.inc

Right here I get errors thrown:

```
depbase=`echo ndpi_network_list_compile.o | sed 's|[^/]*$|.deps/&|;s|\.o$||'`;\
gcc -DHAVE_CONFIG_H -I. -I../../src/include  -I../../src/include/  -I../../src/lib/third_party/include/   -g -O2 -fPIC -DPIC -DNDPI_LIB_COMPILATION  -MT ndpi_network_list_compile.o -MD -MP -MF $depbase.Tpo -c -o ndpi_network_list_compile.o ndpi_network_list_compile.c &&\
mv -f $depbase.Tpo $depbase.Po
In file included from ../../src/include/ndpi_main.h:34,
                 from ../../src/include/ndpi_api.h:73,
                 from ndpi_network_list_compile.c:10:
../../src/include/ndpi_protocols.h:31:1: error: unknown type name ‘NDPI_STATIC’
   31 | NDPI_STATIC ndpi_port_range* ndpi_build_default_ports_range(ndpi_port_range *ports,
      | ^~~~~~~~~~~
../../src/include/ndpi_protocols.h:31:28: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘*’ token
   31 | NDPI_STATIC ndpi_port_range* ndpi_build_default_ports_range(ndpi_port_range *ports,
      |                            ^
../../src/include/ndpi_protocols.h:38:1: error: unknown type name ‘NDPI_STATIC’
   38 | NDPI_STATIC ndpi_port_range* ndpi_build_default_ports(ndpi_port_range *ports,
      | ^~~~~~~~~~~
../../src/include/ndpi_protocols.h:38:28: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘*’ token
   38 | NDPI_STATIC ndpi_port_range* ndpi_build_default_ports(ndpi_port_range *ports,
      |                            ^
../../src/include/ndpi_protocols.h:46:1: error: unknown type name ‘NDPI_STATIC’
   46 | NDPI_STATIC u_int ndpi_search_tcp_or_udp_raw(struct ndpi_detection_module_struct *ndpi_struct,
      | ^~~~~~~~~~~
../../src/include/ndpi_protocols.h:46:19: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘ndpi_search_tcp_or_udp_raw’
   46 | NDPI_STATIC u_int ndpi_search_tcp_or_udp_raw(struct ndpi_detection_module_struct *ndpi_struct,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

Is the INSTALL uptodate? https://github.com/mirror/dd-wrt/blob/master/src/router/ndpi-netfilter/ndpi-netfilter/INSTALL
Is there an easier way to achieve compile it?

I would really appreciate your advice on this matter,

Thanks

houmie, 2020-09-30:
Running ```make``` inside dd-wrt/tree/master/src/router/ndpi-netfilter/ndpi-netfilter also leads to errors:

```
root@de-fsn-x:/home/ubuntu/ndpi-netfilter/ndpi-netfilter# make
gcc -DNDPI_STATIC= -I ../src/include -I ../src/lib/third_party/include -o ../src/lib/ndpi_network_list_compile ../src/lib/ndpi_network_list_compile.c
../src/lib/ndpi_network_list_compile -o ../src/lib/ndpi_network_list.c.inc ../src/lib/ndpi_network_list_std.yaml ../src/lib/ndpi_network_list_tor.yaml
Warning: line   78: '52.82.0.0/14' is not network
Warning: line  423: '141.101.88.0/19' is not network
Warning: line  609: '209.148.214.135/21' is not network
Warning: line  718: '208.64.201.0/22' is not network
Warning: line  724: '159.122.189.48/21' is not network
Warning: line  796: '1.44.48.64/23' is not network
Warning: line  797: '1.44.69.80/25' is not network
Warning: line  798: '2.46.138.52/28' is not network
Warning: line  799: '2.47.103.32/23' is not network
make -C src
make[1]: Entering directory '/home/ubuntu/ndpi-netfilter/ndpi-netfilter/src'
grep: /include/linux/skbuff.h: No such file or directory
$xt_ndpi-y is [main.o] -I/../../src/include -I/../../src/lib -I/../libre -I/../../src/lib/third_party/include  -DHAVE_CONFIG_H -DNDPI_LIB_COMPILATION -DOPENDPI_NETFILTER_MODULE -DNDPI_STATIC=static -w -femit-struct-debug-detailed=any
\
make  -C  M=$PWD modules -w;
make[2]: Entering directory '/home/ubuntu/ndpi-netfilter/ndpi-netfilter/src'
make[2]: *** M=/home/ubuntu/ndpi-netfilter/ndpi-netfilter/src: No such file or directory.  Stop.
make[2]: Leaving directory '/home/ubuntu/ndpi-netfilter/ndpi-netfilter/src'
make[1]: *** [Makefile:26: modules] Error 2
make[1]: Leaving directory '/home/ubuntu/ndpi-netfilter/ndpi-netfilter/src'
make: *** [Makefile:4: all] Error 2
```

I must be missing something, but I went through all READMEs I could find there. :-)

BrainSlayer, 2022-08-24:
from what i see you are missing significant kernel headers or your kernel header location is different and not placed in /usr/include. in that case you need to edit ndpi-netfilter/src/Makefile (see LINUXDIR env in it) to point to the correct kernel header location

closed 2022-08-24 by BrainSlayer

houmie, 2022-09-20:
Hi @BrainSlayer 

I thought I give this another go on Debian 11.  When running ./autoconf  it fails saying that ```config.status: error: cannot find input file: Makefile.in```

It seems a number of files are missing here:
```
configure.ac:336: error: required file 'example/Makefile.in' not found
configure.ac:336: error: required file 'tests/Makefile.in' not found
configure.ac:336: error: required file 'tests/unit/Makefile.in' not found
configure.ac:336: error: required file 'tests/performance/Makefile.in' not found
configure.ac:336: error: required file 'tests/dga/Makefile.in' not found
configure.ac:336: error: required file 'rrdtool/Makefile.in' not found
configure.ac:336: error: required file 'influxdb/Makefile.in' not found
configure.ac:337: error: required file 'tests/do.sh.in' not found
Makefile.am:2: error: required directory ./tests does not exist
```

In example I can see ```Makefile.dpdk.in``` but not the file it's looking for.

Any suggestions?

**Full log:**

```
admin@de-vpn-1:~/dd-wrt/src/router/ndpi-netfilter$ ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:54: installing './compile'
configure.ac:6: installing './missing'
configure.ac:336: error: required file 'example/Makefile.in' not found
configure.ac:336: error: required file 'tests/Makefile.in' not found
configure.ac:336: error: required file 'tests/unit/Makefile.in' not found
configure.ac:336: error: required file 'tests/performance/Makefile.in' not found
configure.ac:336: error: required file 'tests/dga/Makefile.in' not found
configure.ac:336: error: required file 'rrdtool/Makefile.in' not found
configure.ac:336: error: required file 'influxdb/Makefile.in' not found
configure.ac:337: error: required file 'tests/do.sh.in' not found
Makefile.am:2: error: required directory ./tests does not exist
fuzz/Makefile.am: installing './depcomp'
autoreconf: automake failed with exit status: 1
./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
[... 86 lines cut ...]
checking netinet/in.h usability... yes
checking netinet/in.h presence... yes
checking for netinet/in.h... yes
checking for stdint.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking for unistd.h... (cached) yes
checking math.h usability... yes
checking math.h presence... yes
checking for math.h... yes
checking float.h usability... yes
checking float.h presence... yes
checking for float.h... yes
checking for sqrt in -lm... no
checking for rrd_fetch_r in -lrrd... no
checking DPDK (used by ndpiReader)... no (missing /home/admin/DPDK)
checking for numa_available in -lnuma... no
checking for pcap_open_live in -lpcap... yes
checking for pthread_setaffinity_np in -lpthread... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: error: cannot find input file: `Makefile.in'
```

Thanks