mirror/dd-wrt #45: nvram settings for regulatory domain don't match GUI

slycordinator, 2021-09-11:
The website configuration allows for setting a singe regulatory domain for wlan0. You cannot set the regdomain for any secondary radio, which is reasonable. But the subsequent nvram settings don't always match this.

Steps to REPRO:
1) Flash with a ddwrt firmware
2) Change the regdomain in the GUI to a different country from its default
3) commit nvram settings
4) See that wlan0_regdomain="$NEW_COUNTRY" and wlan1_regdomain="$OLD_COUNTRY"

In my GUI:
"Regulatory Domain = KOREA_REPUBLIC" [because I live in South Korea (Republic of Korea)]

NVRAM SETTINGS:
wlan0_regdomain=KOREA_REPUBLIC
wlan1_regdomain=UNITED_STATES

ACTUAL RESULTS:
regdomains for wlan0 and wlan1 are different from each other

EXPECTED RESULTS:
1) There be only one variable, since it's redundant to store multiple settings that are intended to be identical.

or

2) The two variables to be set identically to each other to both match the GUI

Device info:
TP-Link ARCHER-C7 v2 (US)

[image]
[image]

BrainSlayer, 2021-09-12:
there is just one setting. only wlan0_regdomain is used

closed 2021-09-12 by BrainSlayer

slycordinator, 2021-09-12:
"there is just one setting. only wlan0_regdomain is used" 

Imo, it seems to default to generating two settings, only one of which is used.

BrainSlayer, 2021-09-12:
the wlan1_default is orphan on obsolete. or lets say on some older platforms it was possible to run 2 different reg domains. but this isnt the case for mac80211 based drivers