mirror/dd-wrt #169: Build instructions

fmuntean, 2022-05-16:
Can someone add some build instructions please?
How can we use this repo to rebuild the binary ?

egc112, 2022-05-16:
Have send you a PM via DDWRT

kernel-panic69, 2022-06-13:
Just an FYI, official toolchains were updated on May 11th/13th: 
https://download1.dd-wrt.com/dd-wrtv2/downloads/toolchains/

tsl0922, 2023-09-15:
For anyone who want to compile dd-wrt from source, I'm sharing my dd-wrt build scripts and patches here (github actions based).

Althrough it's mainy for PHICOMM K2P (Ralink MT7621 based), should also be useful to others.

https://github.com/tsl0922/DD-WRT

> **NOTE:** The Makefiles and patches are highly optimized for K2P, so don't expect it to work for your router without modification.