mirror/dd-wrt #381: autoconf fails with the error message "cannot find input file: Makefile.in"

houmie, 2022-09-20:
Hi @BrainSlayer 

I thought I give this another go on **Debian 11**.  
When running ./autoconf  it fails saying that ```config.status: error: cannot find input file: Makefile.in```

It seems a number of files are missing here:
```
configure.ac:336: error: required file 'example/Makefile.in' not found
configure.ac:336: error: required file 'tests/Makefile.in' not found
configure.ac:336: error: required file 'tests/unit/Makefile.in' not found
configure.ac:336: error: required file 'tests/performance/Makefile.in' not found
configure.ac:336: error: required file 'tests/dga/Makefile.in' not found
configure.ac:336: error: required file 'rrdtool/Makefile.in' not found
configure.ac:336: error: required file 'influxdb/Makefile.in' not found
configure.ac:337: error: required file 'tests/do.sh.in' not found
Makefile.am:2: error: required directory ./tests does not exist
```

In example I can see ```Makefile.dpdk.in``` but not the file it's looking for.

Any suggestions?

**Full log:**

```
admin@de-vpn-1:~/dd-wrt/src/router/ndpi-netfilter$ ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:54: installing './compile'
configure.ac:6: installing './missing'
configure.ac:336: error: required file 'example/Makefile.in' not found
configure.ac:336: error: required file 'tests/Makefile.in' not found
configure.ac:336: error: required file 'tests/unit/Makefile.in' not found
configure.ac:336: error: required file 'tests/performance/Makefile.in' not found
configure.ac:336: error: required file 'tests/dga/Makefile.in' not found
configure.ac:336: error: required file 'rrdtool/Makefile.in' not found
configure.ac:336: error: required file 'influxdb/Makefile.in' not found
configure.ac:337: error: required file 'tests/do.sh.in' not found
Makefile.am:2: error: required directory ./tests does not exist
fuzz/Makefile.am: installing './depcomp'
autoreconf: automake failed with exit status: 1
./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for cos in -lm... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking whether gcc and cc understand -c and -o together... yes
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for ar... (cached) ar
Setting API version to 13913
checking for json-c... no
configure: WARNING: JSON-C not available. Disabled unit test.
checking for json_object_put in -ljson-c... no
configure: WARNING: JSON-C not available. Disabled unit test.
checking whether gcc is Clang... no
checking whether pthreads work with "-pthread" and "-lpthread"... yes
checking for joinable pthread attribute... PTHREAD_CREATE_JOINABLE
checking whether more special flags are required for pthreads... no
checking for PTHREAD_PRIO_INHERIT... yes
checking netinet/in.h usability... yes
checking netinet/in.h presence... yes
checking for netinet/in.h... yes
checking for stdint.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking for unistd.h... (cached) yes
checking math.h usability... yes
checking math.h presence... yes
checking for math.h... yes
checking float.h usability... yes
checking float.h presence... yes
checking for float.h... yes
checking for sqrt in -lm... no
checking for rrd_fetch_r in -lrrd... no
checking DPDK (used by ndpiReader)... no (missing /home/admin/DPDK)
checking for numa_available in -lnuma... no
checking for pcap_open_live in -lpcap... yes
checking for pthread_setaffinity_np in -lpthread... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: error: cannot find input file: `Makefile.in'
```

Thanks

BrainSlayer, 2022-12-22:
dd-wrt build does not make use of configure. check the npdi-netfilter/Makefile how its made. the Makefiles are hand customized. you may need to adjust them for debian compiling. do not use configure

houmie, 2023-02-19:
@BrainSlayer 

Thank you. I have tried to compile it with `make` only.

This is my current folder structure:

```
root@debian-2gb-nbg1-1:~/ndpi-netfilter# pwd
/root/ndpi-netfilter
```

And the content:
```
root@debian-2gb-nbg1-1:~/ndpi-netfilter# ls
CHANGELOG.md  CONTRIBUTING.md  COPYING	depcomp  doc  example  fuzz  influxdb  INSTALL	lgtm.yml  libndpi.pc.in  m4  Makefile  Makefile.am  ndpi-netfilter  packages  python  README.fuzzer.md	README.md  rrdtool  sonar-project.properties  src  utils  windows
```

So when I run `make` I get this error:

```
root@debian-2gb-nbg1-1:~/ndpi-netfilter# make
make -C ndpi-netfilter
make[1]: Entering directory '/root/ndpi-netfilter/ndpi-netfilter'
gcc -DNDPI_STATIC= -I ../src/include -I ../src/lib/third_party/include -o ../src/lib/ndpi_network_list_compile ../src/lib/ndpi_network_list_compile.c
In file included from ../src/lib/ndpi_network_list_compile.c:10:
../src/include/ndpi_api.h:1465:15: warning: data definition has no type or storage class
 1465 |   NDPI_STATIC ndpi_serialize_binary_double(ndpi_serializer *_serializer, const char *key, u_int16_t klen, double value, const char *format /* e.f. "%.2f" */);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/include/ndpi_api.h:1465:15: warning: type defaults to ‘int’ in declaration of ‘ndpi_serialize_binary_double’ [-Wimplicit-int]
../src/lib/ndpi_network_list_compile -o ../src/lib/ndpi_network_list.c.inc ../src/lib/ndpi_network_list_amazon_aws.yaml ../src/lib/ndpi_network_list_cloudflare.yaml ../src/lib/ndpi_network_list_microsoft_365.yaml ../src/lib/ndpi_network_list_microsoft_azure.yaml ../src/lib/ndpi_network_list_mining.yaml ../src/lib/ndpi_network_list_ms_one_drive.yaml ../src/lib/ndpi_network_list_ms_outlook.yaml ../src/lib/ndpi_network_list_other.yaml ../src/lib/ndpi_network_list_skype_teams.yaml ../src/lib/ndpi_network_list_tor.yaml ../src/lib/ndpi_network_list_whatsapp.yaml ../src/lib/ndpi_network_list_zoom.yaml
Warning: line   55: '239.255.102.18/32' is multicast address!
make -C src
make[2]: Entering directory '/root/ndpi-netfilter/ndpi-netfilter/src'
grep: /include/linux/skbuff.h: No such file or directory
$xt_ndpi-y is [main.o ../../src/lib/ndpi_geoip.o ] -I/../../src/include -I/../../src/lib -I/../libre -I/../../src/lib/third_party/include  -DHAVE_CONFIG_H -DNDPI_LIB_COMPILATION -DOPENDPI_NETFILTER_MODULE -DNDPI_DETECTION_SUPPORT_IPV6 -DNDPI_STATIC=static -Wno-unused-function -Wno-declaration-after-statement
\
make  -C  M=$PWD modules -w;
make[3]: Entering directory '/root/ndpi-netfilter/ndpi-netfilter/src'
make[3]: *** M=/root/ndpi-netfilter/ndpi-netfilter/src: No such file or directory.  Stop.
make[3]: Leaving directory '/root/ndpi-netfilter/ndpi-netfilter/src'
make[2]: *** [Makefile:33: modules] Error 2
make[2]: Leaving directory '/root/ndpi-netfilter/ndpi-netfilter/src'
make[1]: *** [Makefile:7: all] Error 2
make[1]: Leaving directory '/root/ndpi-netfilter/ndpi-netfilter'
make: *** [Makefile:2: all] Error 2
```

What am I missing please?

BrainSlayer, 2023-02-19:
as i said. the Makefiles are hand crafted for dd-wrt use only. you need to modify them to use them in debian. in your case the kernel source path is simply wrong.  see KERNEL_DIR / LINUXDIR in Makefile. this is missing in your host system. so you need to define it. LINUXDIR must point to your kernel source path

houmie, 2023-02-19:
Thank you. I got a step further:

I modified `KERNEL_DIR` inside `src/router/ndpi-netfilter/ndpi-netfilter/src/Makefile`

```
KERNEL_DIR := /lib/modules/$(shell uname -r)/build
```

And now it worked better than last time, but it still errors out.

cd ~/dd-wrt/src/router/ndpi-netfilter

```
root@debian-2gb-fsn1-1:~/dd-wrt/src/router/ndpi-netfilter# make
make -C ndpi-netfilter
make[1]: Entering directory '/root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter'
gcc -DNDPI_STATIC= -I ../src/include -I ../src/lib/third_party/include -o ../src/lib/ndpi_network_list_compile ../src/lib/ndpi_network_list_compile.c
In file included from ../src/lib/ndpi_network_list_compile.c:10:
../src/include/ndpi_api.h:1465:15: warning: data definition has no type or storage class
 1465 |   NDPI_STATIC ndpi_serialize_binary_double(ndpi_serializer *_serializer, const char *key, u_int16_t klen, double value, const char *format /* e.f. "%.2f" */);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/include/ndpi_api.h:1465:15: warning: type defaults to ‘int’ in declaration of ‘ndpi_serialize_binary_double’ [-Wimplicit-int]
../src/lib/ndpi_network_list_compile -o ../src/lib/ndpi_network_list.c.inc ../src/lib/ndpi_network_list_amazon_aws.yaml ../src/lib/ndpi_network_list_cloudflare.yaml ../src/lib/ndpi_network_list_microsoft_365.yaml ../src/lib/ndpi_network_list_microsoft_azure.yaml ../src/lib/ndpi_network_list_mining.yaml ../src/lib/ndpi_network_list_ms_one_drive.yaml ../src/lib/ndpi_network_list_ms_outlook.yaml ../src/lib/ndpi_network_list_other.yaml ../src/lib/ndpi_network_list_skype_teams.yaml ../src/lib/ndpi_network_list_tor.yaml ../src/lib/ndpi_network_list_whatsapp.yaml ../src/lib/ndpi_network_list_zoom.yaml
Warning: line   55: '239.255.102.18/32' is multicast address!
make -C src
make[2]: Entering directory '/root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src'
grep: /include/linux/skbuff.h: No such file or directory
$xt_ndpi-y is [main.o ../../src/lib/ndpi_geoip.o ] -I/../../src/include -I/../../src/lib -I/../libre -I/../../src/lib/third_party/include  -DHAVE_CONFIG_H -DNDPI_LIB_COMPILATION -DOPENDPI_NETFILTER_MODULE -DNDPI_DETECTION_SUPPORT_IPV6 -DNDPI_STATIC=static -Wno-unused-function -Wno-declaration-after-statement
\
make  -C /lib/modules/5.10.0-19-amd64/build M=$PWD modules -w;
make[3]: Entering directory '/usr/src/linux-headers-5.10.0-19-amd64'
grep: /include/linux/skbuff.h: No such file or directory
  CC [M]  /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/main.o
In file included from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/third_party/src/gcrypt_light.c:66,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/protocols.c:8,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/ndpi_main.c:121,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/main.c:3340:
/root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/third_party/src/gcrypt/digest.c:117:1: error: conflicting types for ‘sha256_init’
  117 | sha256_init(sha256_t *p)
      | ^~~~~~~~~~~
In file included from /usr/src/linux-headers-5.10.0-19-common/include/linux/filter.h:24,
                 from /usr/src/linux-headers-5.10.0-19-common/include/net/sock.h:59,
                 from /usr/src/linux-headers-5.10.0-19-common/include/linux/tcp.h:19,
                 from /usr/src/linux-headers-5.10.0-19-common/include/linux/ipv6.h:87,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/main.c:39:
/usr/src/linux-headers-5.10.0-19-common/include/crypto/sha.h:136:20: note: previous definition of ‘sha256_init’ was here
  136 | static inline void sha256_init(struct sha256_state *sctx)
      |                    ^~~~~~~~~~~
In file included from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/third_party/src/gcrypt_light.c:66,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/protocols.c:8,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/ndpi_main.c:121,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/main.c:3340:
/root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/third_party/src/gcrypt/digest.c:200:1: error: conflicting types for ‘sha256_update’
  200 | sha256_update(sha256_t *p, const unsigned char *data, size_t size)
      | ^~~~~~~~~~~~~
In file included from /usr/src/linux-headers-5.10.0-19-common/include/linux/filter.h:24,
                 from /usr/src/linux-headers-5.10.0-19-common/include/net/sock.h:59,
                 from /usr/src/linux-headers-5.10.0-19-common/include/linux/tcp.h:19,
                 from /usr/src/linux-headers-5.10.0-19-common/include/linux/ipv6.h:87,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/main.c:39:
/usr/src/linux-headers-5.10.0-19-common/include/crypto/sha.h:148:6: note: previous declaration of ‘sha256_update’ was here
  148 | void sha256_update(struct sha256_state *sctx, const u8 *data, unsigned int len);
      |      ^~~~~~~~~~~~~
In file included from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/third_party/src/gcrypt_light.c:66,
                 from /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src/../../src/lib/protocols.c:8,
[... cut ...]
make[4]: *** [/usr/src/linux-headers-5.10.0-19-common/Makefile:1856: /root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src] Error 2
make[3]: *** [/usr/src/linux-headers-5.10.0-19-common/Makefile:185: __sub-make] Error 2
make[3]: Leaving directory '/usr/src/linux-headers-5.10.0-19-amd64'
make[2]: *** [Makefile:33: modules] Error 2
make[2]: Leaving directory '/root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter/src'
make[1]: *** [Makefile:7: all] Error 2
make[1]: Leaving directory '/root/dd-wrt/src/router/ndpi-netfilter/ndpi-netfilter'
make: *** [Makefile:2: all] Error 2
```

BrainSlayer, 2023-02-19:
this issue is simply kernel compatiblity

houmie, 2023-02-23:
Thanks, as you suggested, I found another net-filter, that works on Debian. I'll close this then.

closed 2023-02-23 by houmie