mirror/dd-wrt #386: Feature request: Tor Snowflake Proxy

Am0rphous, 2022-09-25:
Hi!
Having the option to support the Tor network in DD-WRT routers are great. However the current setup may create a certain amount of network traffic.

Copying info about Snowflake Proxy from Tor's web site: _"Snowflake is a system that allows people from all over the world to access censored websites and applications. Similar to how VPNs assist users in getting around Internet censorship, Snowflake helps you avoid being noticed by Internet censors by making your Internet activity appear as though you're using the Internet for a regular video or voice call."_

It would be awesome to be able to enable Snowflake proxy within the DD-WRT GUI to help censorship.

- [Compiling from source](https://community.torproject.org/relay/setup/snowflake/standalone/) seems rather simple:
````
git clone https://git.torproject.org/pluggable-transports/snowflake.git
cd snowflake/proxy
go build
nohup ./proxy &
````

Would this be complicated to implement into DD-WRT?

Best regards

egc112, 2022-09-26:
I think it also needs Go so I do not see it happen

BrainSlayer, 2022-09-26:
its written in go. it cannot be used in embedded system. go requires very fat sized runtime libraries and does not work on small devices

closed 2022-09-26 by BrainSlayer