mirror/dd-wrt #444: Netgear R9000 bricks after 4-5-6 newest releases

Am0rphous, 2024-06-23:
Hi!

I  have a Netgear R9000 router and tried updating to the latest version (06-19-2024-r56941), but the router somehow bricks. Wireless networks wont start and several lights wont come on (probably because services aren't starting correct). Trying to reboot and do several resets (5-7) magically makes the wifi work but reset the whole router back with default dd-wrt settings. I assumed downgrading to earlier firmware would give more stability, but I experience the problems with several releases such as:
- 05-07-2024-r56237
- 05-13-2024-r56359
- 05-14-2024-r56380
- 05-24-2024-r56490
- 06-08-2024-r56721 
- 06-12-2024-r56820
- 06-19-2024-r56941

Somehow I got the 05-02-2024-r56182 to make it work.

With Firmware: v3.0-r56941 std (06/19/24)
Running `dmesg|grep -E 'error.*failed|failed.*error'` gives
````
bq32k: probe of 0-0068 failed with error -5
ath10k_pci 0001:03:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
ath10k_pci 0001:04:00.0: Direct firmware load for ath10k/QCA9984/hw1.0/firmware-6.bin failed with error -2
````
I was able to make the latest release work 1 time and ran `dmesg` and got the output below. Not sure if it is helpful.
````
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux (LTO) version 4.9.337-rt197 (root@server2) (gcc version 14.1.0 (OpenWrt GCC 14.1.0 r26085-2aa6c875f8) ) #744 SMP Wed Jun 5 03:00:10 +07 2024
[    0.000000] CPU: ARMv7 Processor [412fc0f4] revision 4 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt:Machine model: Annapurna Labs Alpine Dev Board
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 262144
[    0.000000] free_area_init_node: node 0, pgdat c0a2f340, node_mem_map ef7f8000
[    0.000000]   Normal zone: 1536 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:31
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 12 pages/cpu s20108 r8192 d20852 u49152
[    0.000000] pcpu-alloc: s20108 r8192 d20852 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260608
[    0.000000] Kernel command line: root=/dev/mtdblock5 rootfstype=squashfs ubi.mtd=netgear pci=pcie_bus_perf console=ttyS0,115200
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1028456K/1048576K available (7168K kernel code, 195K rwdata, 664K rodata, 1024K init, 390K bss, 20120K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0800000   (8160 kB)
[    0.000000]       .init : 0xc0900000 - 0xc0a00000   (1024 kB)
[    0.000000]       .data : 0xc0a00000 - 0xc0a30e48   ( 196 kB)
[    0.000000]        .bss : 0xc0a32000 - 0xc0a93a79   ( 391 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] Registering 64 msixs, starting at 96
[    0.000000] 	Offload RCU callbacks from all CPUs
[    0.000000] 	Offload RCU callbacks from CPUs: 0-3.
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
[    0.000003] sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
[    0.000009] Switching to timer-based delay loop, resolution 20ns
[    0.000064] clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5096694524 ns
[    0.000074] sched_clock: 32 bits at 375MHz, resolution 2ns, wraps every 5726622718ns
[    0.000180] Calibrating delay loop (skipped), value calculated using timer frequency.. 100.00 BogoMIPS (lpj=500000)
[    0.000189] pid_max: default: 32768 minimum: 301
[... 541 lines cut ...]
[   93.363955] ath10k_pci 0001:03:00.0: Init Channel Number 40
[   93.394573] ath10k_pci 0001:03:00.0: set base mac addr 78:d2:94:58:6a:80
[   93.434489] br0: port 4(wlan0) entered blocking state
[   93.439551] br0: port 4(wlan0) entered disabled state
[   93.444790] device wlan0 entered promiscuous mode
[   93.461196] ath10k_pci 0001:03:00.0: mac txpower 20
[   93.732564] ath10k_pci 0001:03:00.0: mac txpower 20
[   93.742705] br0: port 4(wlan0) entered blocking state
[   93.747752] br0: port 4(wlan0) entered forwarding state
[   93.804748] ath10k_pci 0001:04:00.0: Init Channel Number 14
[   93.810871] ath10k_pci 0001:03:00.0: Init Channel Number 91
[   94.076299] ath10k_pci 0001:04:00.0: Init Channel Number 11
[   94.082268] ath10k_pci 0001:03:00.0: Init Channel Number 173
[   94.100325] ath10k_pci 0001:03:00.0: Init Channel Number 40
[   94.109491] ath10k_pci 0001:04:00.0: Init Channel Number 11
[   94.212022] device vlan2 entered promiscuous mode
[   94.735966] NET: Registered protocol family 10
[   95.134256] device eth2 entered promiscuous mode
[   95.139240] device vlan2 left promiscuous mode
[   95.145002] device eth2 left promiscuous mode
````

Let me know if there is something I can do to help identify the issue.

Best regards

kernel-panic69, 2024-06-23:
This repository isn't for seeking help, reporting bugs, etc.; submitting patches is welcome, though.

https://github.com/mirror/dd-wrt/blob/master/README.md

But don't open a ticket, open a query in the forum:

https://forum.dd-wrt.com/phpBB2/viewforum.php?f=28

Keep in mind that these devices are known for the radios failing due to excessive heat. So, to rule that out, you must test on stock firmware.  Nobody else is reporting these issues in the forum.

BrainSlayer, 2024-08-10:
your log shows that all is working and i own a r9000 which is flashed with latest releases which is working too. so i dont see any issue here

BrainSlayer, 2024-08-10:
DD-WRT v3.0-r57447 std (c) 2024 NewMedia-NET GmbH
Release: 07/17/24
Board: Netgear Nighthawk X10
==========================================================

     ___  ___     _      _____  ______       ____  ___
    / _ \/ _ \___| | /| / / _ \/_  __/ _  __|_  / / _ \
   / // / // /___/ |/ |/ / , _/ / /   | |/ //_ <_/ // /
  /____/____/    |__/|__/_/|_| /_/    |___/____(_)___/

                       DD-WRT v3.0
                   https://www.dd-wrt.com

==========================================================

BusyBox v1.36.1 (2024-07-17 06:29:21 +07) built-in shell (ash)

root@ALPINE:~# softwarerevision
57447
root@ALPINE:~# nvram get DD_BOARD
Netgear Nighthawk X10
root@ALPINE:~#

Am0rphous, 2024-08-14:
Hi, thanks for reply!
I tried for 4 hours to update to the latest firmware but the router goes into reboot-loop. I see it failes 5 times before everything is reset with default dd-wrt wifi name. This is because the config states if 5 failed boot attempts occurs, then reset. My issue therefore seem to be the backup of the configuration file. Any issues related to importing configuration files?

I successfully installed an old firmware version and also successfully imported my config file. Updating to new firmware seems to crash due to my config, even though it is not that complicated. I consider do a write down of config and setting up things from scratch.

BrainSlayer, 2024-08-15:
the only advise i can give. try to find out the conflicting setting or use a serial log and send me the output. so i can check whats happening

Am0rphous, 2024-08-15:
Thanks for tip! I tried reading in the doc for the R9000 but couldn't find anything about serial or console port. But behind the router it states "serial" I assume at the number 1 RJ-45 port. Is that the serial/console port?

kernel-panic69, 2024-08-15:
No, the serial port is on the mainboard inside the router. You have to disassemble to some degree to get to it:

https://i.imgur.com/cPBEwxy.jpeg

https://forum.dd-wrt.com/phpBB2/viewtopic.php?p=1214461#1214461

BrainSlayer, 2025-04-19:
closed due inactivity

closed 2025-04-19 by BrainSlayer