mirror/dd-wrt #453: DD-WRT includes a backdoored version of xz-utils versioned 5.6.1

MarioNooo, 2025-01-07:
Hello, I've noticed that you include a vulnerable version of xz with the CVE identifier CVE-2024-3094

Please update it immediately.

More info here: https://en.wikipedia.org/wiki/XZ_Utils_backdoor

egc112, 2025-01-08:
What DDWRT version are you talking about?

kernel-panic69, 2025-01-08:
https://github.com/mirror/dd-wrt/blob/master/src/router/xz/NEWS

Current release is 5.63:

https://github.com/tukaani-project/xz/blob/master/NEWS

kernel-panic69, 2025-01-08:
https://github.com/mirror/dd-wrt/blob/master/src/router/xz/NEWS
Current version is 5.6.3:
https://github.com/tukaani-project/xz/blob/master/NEWS

BrainSlayer, 2025-01-09:
> Hello, I've noticed that you include a vulnerable version of xz with the CVE identifier [CVE-2024-3094](https://github.com/advisories/GHSA-rxwq-x6h5-x525)
> 
> Please update it immediately.
> 
> More info here: https://en.wikipedia.org/wiki/XZ_Utils_backdoor

no i wont. the backdoored version was only published on the website but never on github. the question is where you obtained the source

closed 2025-01-09 by BrainSlayer

MarioNooo, 2025-01-09:
Okay I understand that now