mirror/dd-wrt #455: ProtonVPN as OpenVPN client is broken since r58070 on my Linksys router

missddwrt, 2025-01-27:
I use ProtonVPN on my Linksys WRT1900ACSv2 router. It worked for years. But since r58070 it does not work anymore. UDP and TCP both are not working. I'm currently using r57956. There have been others reporting the same issue, they gave up and switched to Wireguard. I can't do that because I need to use TCP. 

messages (verb 6): https://pastebin.com/raw/NnhpNNYM
openvpn.conf: https://pastebin.com/raw/urkKdNp4

I made a topic about it on the forum: https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=337561

https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=337985 - here someone with another VPN assumes OpenSSL caused the issue. If I remember right r58070 has an OpenSSL update too (I cannot confirm that because svn.dd-wrt.com is broken for me)

My config is the same as the original ProtonVPN DDWRT instructions. A couple of months ago they changed the DDWRT instructions and tell their users to import the .conf file https://protonvpn.com/support/vpn-router-ddwrt/. The config is the same and doesn't solve the issue

ProtonVPN has a free plan. The exact same config from the ProtonVPN website has the same issue. It's easy to reproduce

kernel-panic69, 2025-01-27:
The following tickets were closed because of lack of information and desire to change anything configuration related. Even the thread you've posted shows a lack of following advice IMHO.

https://svn.dd-wrt.com/ticket/7793

https://svn.dd-wrt.com/ticket/7796

Perhaps sir egc will see this and comment as well - or perhaps you need to re-visit the thread.

egc112, 2025-01-31:
It is very well possible that OpenVPN is broken on these linksys targets maybe `cryptodev` problem on these specific targets.
But I do not have a Linksys target so cannot test

missddwrt, 2025-02-08:
@kernel-panic69 
"Even the thread you've posted shows a lack of following advice IMHO."
Hi guys. Following what advice? I did not take Alozaros serious because he tells me to set verb6, while he could easily see I have logging already on verb6. Neither did his recommendations solve the issue. 

I'm willing to follow all instructions you give me to debug this issue. Next weekend I have time to give it a shot

kernel-panic69, 2025-02-09:
@missddwrt this is still not the place to discuss. Continue discussion / open a thread in the forum. Thanks.

BrainSlayer, 2025-04-19:
i close this here since due no response. user uses likelly a wrong openvpn configuration since config syntax in newer openvpn version s might has changed

closed 2025-04-19 by BrainSlayer